chirimen-oh/chirimen #2: Check with Firefox (not only chromium)

dynamis, 2017-11-02:
Current examples are prepared for / tested with Chromium, default browser of RPi3. But technically no difficulties with Firefox I guess. It's better if we can test/confirm with Firefox too.

This need not to be official support of the project. Just want to check and share the result (on wiki or somewhere).

satakagi, 2018-09-13:
To install firefox-esr:
https://www.raspberrypi.org/forums/viewtopic.php?t=188037

`sudo apt-get update`
`sudo apt-get install firefox-esr`

satakagi, 2018-09-14:
Due to security restrictions of firefox, access to `wss://localhost:33330/` is initially restricted.
However, you can set the exception by entering `https://localhost:33330/` in the URL input field of firefox. Then, the above problem was solved.

I was able to verify examples of GPIO (Blink and Button), I2C (ADT7410 and S11059), and running on JSBin as well as Local will work well.

closed 2018-09-14 by satakagi