chirimen-oh/chirimen #9: イベント用にテキストエディタ入りのイメージが必要

gurezo, 2017-12-17:
### subject
イベントなどで、Raspbianに別途テキストエディタ（標準以外）を入れた
イメージを用意したほうが良い。
※ [はじめての JavaScript 1DAY 講習会 ](https://connpass.com/event/72869/)からのフィードバック
  WindowsとmacOSとで、Javascriptの実習で、フォローが必要だった為

### 候補
- atom
- sublimetext
- vscode
-  any

### TODO
- エディタの選定
- イメージの作成
- 配布物の変更（ダウンロードファイル）

tadfmac, 2017-12-21:
CHIRIMEN for Raspiでは基本的に、jsfiddleとかjsbinを使う方針なんですが、それではだめですか?

gurezo, 2017-12-26:
@tadfmac issue を立てる場所を間違えたようです。
> ※ [はじめての JavaScript 1DAY 講習会 ](https://connpass.com/event/72869/)からのフィードバック
>   WindowsとmacOSとで、Javascriptの実習で、フォローが必要だった為
Javascriptの実習をやるPCが、WindowsとmacOSとで、異なるサポートが必要だったのが問題でした。
Javascriptの実習をPC行う際にどちらかのOSに統一しましょうというのが、命題でした。

gurezo, 2017-12-26:
issue 違いなので、closeします。

closed 2017-12-26 by gurezo