chirimen-oh/chirimen #16: なぞのimage

tadfmac, 2018-08-10:
https://github.com/chirimen-oh/chirimen-raspi3/wiki
を見ると、

> OS Image - 日付を見て最新のイメージを選択しダウンロード後、下のセットアップガイドに従ってください

と記載されている。

上記記述に従うと、現在、`CHIRIMEN_Raspberry_pi_3_20180716.zip` が最新版のように見える。
しかし、現在 [Release](https://github.com/chirimen-oh/chirimen-raspi3/releases) を見ると、[Ver. 2018/03/08](https://github.com/chirimen-oh/chirimen-raspi3/releases/tag/20180308) が最新版のように見える。

このような状況は混乱を招くし、そもそも`CHIRIMEN_Raspberry_pi_3_20180716.zip`が何を改変し何を目的として置かれたimageなのか不明。したがって下記を明確にする必要があると考える。

- [ ] `CHIRIMEN_Raspberry_pi_3_20180716.zip`の前Ver.との差分 (従来通り Raspberry Pi 3 Model B上で動作するイメージなのか? すらわからない)
- [ ] [Release](https://github.com/chirimen-oh/chirimen-raspi3/releases) への反映 (release時はreadme.mdのversion表記やrelease.shで生成されるVerssionを更新する必要がある)

dynamis, 2018-08-13:
リリース手順がしっかりしてないと混乱の元だし、ビルドイメージの違い (内容およびテスト状況=正式リリースかどうか) が分かるようにリリースノートなり changelog なりを書きましょうって話ですよね。

tadfmac, 2018-08-15:
普通にダウンロードして使うだけの人にとって、現状ちゃんと判る人居る？って話です。