chirimen-oh/chirimen #17: PCA 9685 の配線図に誤りがある

dynamis, 2018-08-12:
電源の配線が間違ってました:
https://github.com/chirimen-oh/chirimen-raspi3/blob/master/gc/i2c/i2c-PCA9685/schematic.png
USBや電池から電源を供給するのは PCA 9685 中央のネジ止めする青いところに刺す感じです。

fritzing で編集すれば良いのかなと思ったのですが、png エクスポート前のオリジナルデータはリポジトリに入っていたりしない感じでしょうか。

tadfmac, 2018-08-13:
Adafruit 製 PCA9685 ボードには複数のV+ の端子があり、現在の図でもV+に接続されており、ご指摘の中央のネジ止め式ターミナルで供給されるのも V+ で共通の認識です。
(ケーブル終端の形状で選べるようこうなってる)

tadfmac, 2018-08-13:
修正する場合、こちらをご利用ください〜。
[PCA9685.fzz.zip](https://github.com/chirimen-oh/chirimen-raspi3/files/2281628/PCA9685.fzz.zip)

dynamis, 2018-08-13:
左右のスルーホールに V+ 繋いでも動かないがネジ止めターミナルに繋いだら動いたので、サーボモータ向けの電力共有はターミナルからだけなのかと思って書いたのですが、本来左右のでも良かったのですね。

確かに製品ページでは Reverse polarity protection があるかないかの違いはあるが、どっちでも良いと書かれてますし配線見る限りどっちでも良いように見えますね。。。
https://www.adafruit.com/product/815
* Terminal block for power input (or you can use the 0.1" breakouts on the side)
* Reverse polarity protection on the terminal block input.

もしかしたら昨日動かないと言ってた時には 5V 入力を V+ じゃなくて VCC とかに繋いでたのかな。。。来週再度試します。。。

tadfmac, 2018-08-14:
もしかしたら昨日動かないと言ってた時には 5V 入力を V+ じゃなくて VCC とかに繋いでたのかな。。。
大量にサーボ繋ぐ時は、強い電力＋強いケーブルが使いたくなるので真ん中のターミナル推奨ですね。
ジャンパーピンとかでやるときは、1コ2コしか繋げないけど、ピン側で電源供給するとやりやすいです。

※ボロいジャンパー線は電流たくさん流すと簡単に燃えるので、そっちの方が注意ですね。