chirimen-oh/chirimen #20: rspi3 ローカルに jsbin をインストールする

dynamis, 2018-08-16:
現在 example コードはその場で編集が可能なように jsbin 等を使っているが、exmaple の修正などを行うときに (pro user アカウントを持たない限り) API での書き換えも出来ずにリリースが非常に煩雑となっていてコミュニティベースで素早く修正していくのが困難であったり、jsbin などのサービスが停止していると使えなくなってしまう問題がある。

jsbin も rpi3 上にローカルインストールしてしまえば example や tutorial のコードは github repo のクローンした物をそのまま自動で読み込んだページを作成したり出来るようになるし、polyfill などの仕様が変わってもバージョン不整合 (非互換) 問題も起きずに動くことが保証できる。既に example 用の node も立ってるし、それくらいしてしまっても良いのではなかろうか。

逆に他のマシンで動作させたりとかが出来なくなることは課題。

dynamis, 2018-08-16:
ref
* jsbin インストール手順 https://jsbin.com/help/running-a-local-copy-of-jsbin/
* jsbin API: https://jsbin.com/help/api/

sizuhiko, 2018-08-16:
ローカルのインストールガイド読みましたが、サポートしているバージョンが `0.10.x` で

> Installing JS Bin on newer versions of Node is not supported and may fail

ということなので、どうやって複数バージョン起動するのかが悩みそう。
Docker使えば良さそうだけど、rpi3 でどうかなぁ？（正式対応はしていますが...
https://github.com/euprogramador/jsbin

dynamis, 2018-08-16:
その docker は起動してみたら 2 年前の jsbin バージョンが古いもので jslint とかの面で使い気持ちになりました。
(jslint で `{ "esnext": true }` とか指定して色々警告でないようにしたい)

一方で raspi3 でなく取りあえず mac のローカルで jsbin インストールしてみた感じでは、node バージョン依存ではなく mysql バージョンのデフォルト設定の違いとか https 対応とかあたりで色々面倒そうな雰囲気を感じました:

* v4 からは sqlite じゃなくて mysql 必要だぜっていきなり怒られたので mysql も用意する
* mysql の DB 名は jsbin でスキーマは build/full-db-v3.mysql.sql を読み込む
* ER_NOT_SUPPORTED_AUTH_MODE って怒られるときは mysql 起動できてないか認証方式の制限
* ER_NO_DEFAULT_FOR_FIELD って怒られるので mysql の設定から STRICT_TRANS_TABLES を削除
  * https://github.com/jsbin/jsbin/issues/2414
* http://localhost:3000 で起動してるのに script は https://localhost:3000/ から読み込もうとするから動作してないｗ

もっとちゃんと調べて最新版を動かせるようになったら。。。ですね。

sizuhiko, 2018-08-17:
3系の最新版は動いたので、

> (jslint で { "esnext": true } とか指定して色々警告でないようにしたい)

だけであれば、ローカル起動する JSHint のデフォルトオプションを書き換えてしまっても良いのかな？と思いました。
`feat/next-v5` もトライしてみたけど、babelでビルドしたjsがエラーになっていて、わからず...（jsbinがデプロイしているv5は動くのになー...）

dynamis, 2018-08-17:
私の方でも feat/next-v5 試しました (on mac)。npm バージョンが古いと npm install 時点で失敗したりしましたが更新してからだと npm install; npm start で起動して  http://localhost:3000/ にで起動しました (セキュリティ警告とか出るのは無視)。

コード見た時点で怪しいなと思ってましたが、全く UI とかが違う超シンプルな React App になってました。HTML/JS/CSS は切り替え表示で、同時表示ではないのがそうきたかーって感じですが、Cmd+1,2,3 で切り替えられるからまぁ慣れれば小さい画面ではいいかも。

[image]

**npm build でプロダクションビルドを作ったものを serve するとこんな感じです:
https://vibrant-bohr-21e5cc.netlify.com/**

最初使い方が分からず (そもそもページの保存の仕方が分からなかった！) 困ったのですが、Cmd+Shift+P で出てくるメニューから色々出来るというか、編集後に単に Cmd+S で https://vibrant-bohr-21e5cc.netlify.com/local/xxxxx-xxxxx-xxx 的なページが作られることが分かりました。

但し実データはブラウザの IndexedDB に保存されるもので、ブラウザローカルではなく publish するのは gist や codepen に連携して保存する形のアーキテクチャに変わるようです。(なんの認証情報も設定してないので当たり前ながら) エラーで出来ませんでしたが、github auth でログインしたら次のことが出来るとか書いてありました:

- Publish and share URLs
- Export to Gists and embed in your website
- Create privately shareable bins
- Sync your settings
- 5GB of hosted assets

## 感想
* オフラインで使えるシンプルな開発環境を提供するというのは満たせそうです
  * ビルドしてしまえば raspi3 上では serve とか http-server とかでホストするだけ
  * JS パネルの lint というか、コンソールのエラーと連動した構文チェックなどもちゃんと動いていて、正しい arrow function とか Promise 構文なのにエラー扱いされることはないのでそこは快適
* データは Chrome に保存しておく形でもまぁ良いが、イメージを作成するときにゼロから作るとするとちょっと面倒 (Chrome のプロファイルデータは既存イメージから持ってくる形で作業手順の確立が必要)
* examples とかを変えたら簡単に反映させるという件については、サーバ実装ではなくなったので (GitHub Auth 動かしてそれと同期させるとかできればそれでいいが) indexeddb に保存されたデータをローカルファイル (git clone したもの) から読み込んで書き換えるアドオンか何かを作る必要があり、API たたいて更新よりちょっと面倒そう

ちなみに jsbin がデプロイしているのは v4 系のハズ (UI, 機能的にも機能テストして一部動いていたものと同一)

dynamis, 2018-08-23:
example ページにバージョニングがなければ上書き保存されてしまう問題がある。
既存の html から新規に作るインポート機能を用意するかバージョニングの対応が必要と思われる。

dynamis, 2019-07-31:
codesandbox な気持ちが高まってきたのでもうこちらはないだろうと close します (いややっぱりローカル jsbin 欲しいという人がいれば reopen してください)

closed 2019-07-31 by dynamis