chirimen-oh/chirimen #24: リリース用テストスイートと自動化スクリプトの作成

dynamis, 2018-08-23:
自動テスト手順を定義していく中で、

- [ ] どのピンで何をテストするのか定義する
  - [ ] IN/OUT に使用する GPIO ピンのリストを決める (サポート予定のピン全部)
  - [ ] I2C でテストするデバイスリストを定義する (基本的には example に入っているもの全部)
- [ ] テストスイートを実行するための配線を定義、実配線したものを用意する
  - [ ] 配線図を定義
  - [ ] 実機サンプルを用意して WebDINO オフィスに保管
- [ ] テスト用の Web ページ群を用意する
- [ ] テスト用 Web ページ群を自動で順番に表示するスクリプトを用意する
  - web driver か pepetter でやると良いはず
- [ ] 使い方・手順説明をリリース手順書の中に書く

dynamis, 2018-08-29:
実際のテスト手順はこんな感じだった:

- 次の写真のように配線をする
  - ...
- i2cdetect -y -r l
  - 次の 13 デバイスを認識する: 29, 2a, 38, 39, 3c, 40, 41, 48, 49, 53, 5a, 73
  - 73 番のデバイスは初期化に時間がかかるので検出しない場合は暫く待ってから再度実行して確認する
- ブラウザで http://localhost に接続し、testSet/testall.html を開く
- nextText ボタンをして一つずつ iframe 内のサンプルコードで GPIO 入出力、I2C センサーを順番に確かめる
  - GPIO の LDE blink のテストページのボタンは start, all 以外はテスト用のではないので使わない
  - タッチセンサーは初期化が遅いので
  - 紫外線センサーは屋内だと 0, 1 とかぐらいしか出ないが気にしない。

satakagi, 2018-09-11:
一般的なブレッドボードなどで出来る配線(GPIO部)
[image]

I2Cのほうは定義するまでもないですよね。

closed 2018-09-11 by satakagi

satakagi, 2018-09-11:
追伸：
I2Cデバイスは、そのままだとアドレスが衝突するものがいくつかあるので、下表のように変更などをしています。

|Device|NativeAddr|ChangedAddr|
|---|---|---|
|ADT7410|	0x48|	=>test brd:0x49|
|ADS1015|	0x48| |
|VEML6070|	0x38, 0x39| |
|S11059|	0x2a| |
|PCA9685|	0x40|	=>test brd:0x41|
|grove-touch|	0x5a| |
|grove-oledDisplay|	0x3c| |
|grove-gesture|	0x73| |
|grove-light|	0x29| |
|grove-accelerometer|	0x53| |
|GP2Y0E03|	0x40| |