chirimen-oh/chirimen #27: 人感センサー exampleの追加

tadfmac, 2018-08-28:
# 課題

GPIOのexampleが現在少なく、拡充が必要な認識です。
いくつか必要なものがありますが、まずはハッカソン等でニーズの多かった、人感センサーのサンプルを追加したいと思います。

※明日、弊社に実習に来てCHIRIMENを学んでいる成田さんに登録をお願いする予定です。

dynamis, 2018-08-28:
note: 弊社オフィスにはこちらの在庫があります＆最新では試してないけど昔慶応の授業時に使ったことがあるはずです:

https://www.marutsu.co.jp/pc/i/238993/

tadfmac, 2018-08-29:
成田さんよりPull Requrst頂きました。
https://github.com/chirimen-oh/chirimen-raspi3/pull/28

# 制限事項

人感センサーは非検出状態がLOWで、検出状態がHIGHとなるため、プルダウンポートを用いる必要があります。現在、プルダウンポートのGPIO監視には、下記不具合があります。

https://github.com/chirimen-oh/chirimen-raspi3/issues/25

このため、上記不具合を直さないと動きません。

dynamis, 2018-08-29:
今回のプルリクで対象とした具体的な人感センサーはこちらとのこと:
http://www.kyohritsu.sakura.ne.jp/prowiki/index.php?%BF%CD%C2%CE%C0%D6%B3%B0%C0%FE%B4%B6%C3%CE%C1%C7%BB%D2%2FKP-IR412

dynamis, 2018-08-29:
マージされたので close しましょう

closed 2018-08-29 by dynamis