chirimen-oh/chirimen #30: i2cdetectをブラウザで確認できるようなものがあるといいですね

satakagi, 2018-08-29:


satakagi, 2019-07-19:
```javascript
<!DOCTYPE html>
<html>

<head>
<meta charset="utf-8" />
<meta http-equiv="X-UA-Compatible" content="IE=edge"/>
<title>I2C Devices Detector</title>
</head>

<style>
table {
  table-layout: fixed;
}

td {
  width: 30px
}
</style>

<script type="text/javascript" src="https://chirimen.org/chirimen-raspi3/gc/polyfill/polyfill.js"></script>

<script>
// i2cDetect的なものを実装してみます
// https://kernel.googlesource.com/pub/scm/utils/i2c-tools/i2c-tools/+/v3.1.2/tools/i2cdetect.c
// によると、 readmodeの場合はi2c_smbus_read_byteしてるだけなんですね・・

var microBitBle;

var i2cPort;

var readEnable;

onload = async function(){
	makeTable();
	var i2cAccess = await navigator.requestI2CAccess();
	i2cPort = i2cAccess.ports.get(1);
	readData();
}

async function readData(){
	for ( var i = 0 ; i < 128 ; i++ ){
		document.getElementById("ADDR"+i).innerText = "";
	}
	for ( var slaveAddress = 0 ; slaveAddress <128 ; slaveAddress++ ){
		var i2cSlave = await i2cPort.open(slaveAddress);
		try{
			var ret = await i2cSlave.readBytes(8);
			// エラーが起きるところはつながってない
//			console.log("addr:",slaveAddress,"  ans:",slaveAddress.toString(16));
			document.getElementById("ADDR"+slaveAddress).innerText = slaveAddress.toString(16);
		} catch ( e ){
			document.getElementById("ADDR"+slaveAddress).innerText = "--";
//			console.log("addr:",slaveAddress,"  ans: ERROR");
		}
		await sleep(10);
	}
}

function makeTable(){
	var tbl = document.createElement("table");
	var addr =0;
	for ( var i = 0 ; i < 9 ; i++ ){
		var tr = document.createElement("tr");
		
		for ( var j = 0 ; j < 17 ; j++ ){
			var td = document.createElement("td");
			if ( i==0 ){
				if ( j==0){
				} else {
					td.innerText= ((j-1)).toString(16);
				}
			} else {
				if ( j==0 ){
					td.innerText= ((i-1) * 16).toString(16);
				} else {
					td.id="ADDR"+addr;
					++ addr;
				}
			}
			tr.appendChild(td);
		}
		tbl.appendChild(tr);
	}
	detect.appendChild(tbl);
}

</script>    

<body>

<input type="button" value="RE Detection" onclick="readData();" /> 

<div id="msg">---</div>
<div id="detect"></div>

</body>
</html>
```

dynamis, 2019-07-31:
よく使うことになりそうなので gc/contrib ではなく gc/i2c に移動、example からのリンク、チュートリアルからのリンクなども済ませた

https://github.com/chirimen-oh/tutorials/issues/47

closed 2019-07-31 by dynamis