chirimen-oh/chirimen #33: [リリース作業効率化] 壁紙画像のバージョン番号の更新を効率化する(もしくはやめる)

forte916, 2018-09-04:
### Issue
* リリースの度に、壁紙画像に埋め込まれているバージョン番号を更新する必要がある
* 現在は手動で更新しており、自動化するには難易度が高い

### Goal
* デスクトップのすぐ見える位置にバージョン番号が表示されていること ---(A)
* (A) の実現を自動化すること

### Solution
* Plan A
    * 頑張って画像を自動生成する
    * ImageMagick 等のツールを駆使して頑張る方向性は、学習コストに見合っていない気がする

* Plan B
    * 画像に埋め込むのをやめ、デスクトップの別の場所にバージョン番号を表示する
    * System tray にバージョン番号を表示するのはどうか?
    * System tray の方が、後々ショートカットを追加できるし拡張性が高い気がする

### See also
* GTK + Python でバッテリー残量を System tray に表示するサンプル
    * [Battery Tray](https://github.com/lentinj/Battery-Tray)

Plan B の方向でサンプルを作る予定

sizuhiko, 2018-09-04:
一応、npmのモジュールで画像に文字を追加することも可能です。
https://github.com/navjotdhanawat/dynamic-watermark

sizuhiko, 2018-09-04:
> なので大きい・重たいライブラリとかはインストールしたくない

あ、なるほど。`dynamic-watermark` は GraphicsMagick か ImageMagick が必要なんで、そうじゃ無い方が良いのかな。

dynamis, 2018-09-13:
Chrome の headless で画像の上に文字おいてスクリーンショット撮ったらかっこいい (妄想だけ)

ある程度手動で良いなら Canvas に画像読み込んで文字を書き込み右クリックで Canvas の画像を保存する的なフローにするとかでも良いかもしれない。これなら比較的簡単に用意できる。

他にもうちょっと Web 的で自動化できる方法がないか考え中...

* Plan C デスクトップが面倒なら、ブラウザで開くホーム画面にバージョンを表示しちゃう。簡単に確認できれば良いと言うことなら。

tadfmac, 2018-09-14:
puppeteer とか超便利なんだけど、npm i で入れると chromiumがもう一つ入るのは、どうにかならんかなー、という悩みが。。。

dynamis, 2018-09-19:
確かにもうひとつ chromium 入れてる場合じゃないですね。
方向性としてはシステムトレイが私も良いと思います。
+1

satakagi, 2018-09-26:
面倒なら、ソリューションが構築できるまでは、壁紙はやめて、代わりにTopページのほうにhtmlで書いておくのはどうでしょう？　これならかんたん。

dynamis, 2018-10-05:
10/5 版でもこれを忘れていました＆気づかなかったです。(^^;
web 記載なら setup.sh に私でも追記して出来そう。

取りあえず一旦壁紙は止める方向で issue タイトル変えました。

satakagi, 2019-07-26:
壁紙はひとまずやめて、簡単に・・・topページのfooterとしてバージョン文字列表示するようにしました。
https://github.com/chirimen-oh/chirimen-raspi3/commit/77ac724c8880cbde2e3b0eabc7c96fdc3a4ec28b

closed 2019-07-26 by satakagi