chirimen-oh/chirimen #45: 2018/9版をリリースする

satakagi, 2018-09-26:


dynamis, 2018-10-02:
10 月になったので 2018/10 月版としてリリースする。具体的には 10/4 (木) の作業で計画中

dynamis, 2018-10-05:
以下の点が少し気になりますが、ひとまずは

- 初回起動時に raspbian update したよという notice が表示される
- chromium 初回起動時にタブを復元したよと言う notice が表示される (以前もしていたと思う。ダメじゃないけど気持ち悪いので何故か要調査)
- h264ify (chromium アドオン) が無効化されてるから再度権限付与して有効化してねって表示される。Chrome のアップデートでもされてバージョン互換でも生じたのだろうか？それとも初回は出るものなのか？

以下は些末な問題と残りの todo な感じですが、ひとまず一式の動作として致命的な問題はないと検証されたのでリリース版とさせて頂ければ:
- jsbin リンク先がまだ用意できていないものはこれから用意する (ビルドにはリダイレクト URL のブックマークが入っているためビルド側は変更不要)
  - rawgit のままのページ達: https://github.com/chirimen-oh/r.chirimen.org/blob/master/public/_redirects
-  壁紙のビルド日付が 9.14 だけど今日は 10.5
  - まぁ。。。いいか。。。動作上の害はないし、9/14 時点の中身でブックマークとかが変わった程度という意味合いでご容赦を。。。

なお、チュートリアルはオンラインにしかないビルドなので実際使う人が現れるまでの間にいつでも更新可能なので、そちらに問題が見つかってもリリースとしては影響ありません。

dynamis, 2018-10-05:
ブックマークミスがあったバージョン: https://download.chirimen.org/release/raspberry_pi_3/CHIRIMEN_Raspberry_pi_3_20181004RC1.zip

それを修正したバージョン:
https://download.chirimen.org/release/raspberry_pi_3/CHIRIMEN_Raspberry_pi_3_20181005.zip

こちらにて一旦リリースと言うことでクローズさせてください。

dynamis, 2018-10-05:
おっと、イメージを公開する以外にタグを付けるとか色々やることがまだある気がするので reopen

dynamis, 2018-10-05:
- [ ] リリースしたコミットにタグを付ける
- [ ] changelog ファイルを用意して書き込む

satakagi, 2018-10-09:
jsbinリンク先、準備できてないのはkanzashi,stepping motor, 人感センサぐらいですかね?

satakagi, 2018-10-09:
あ、操作ミス再オープンします。

closed 2019-07-31 by dynamis