chirimen-oh/chirimen #47: jsbinのexamplesのpolyfill.js及びデバイスドライバ.jsの参照先を決める

satakagi, 2018-09-28:
いずれも今のところはmz4u.netになっており、リリース作業は自動化されているわけではないようです。
jsbjn内のコードなのでこのリポジトリが直接影響を受けるわけではないですが、ここでリリースされたコードが自動的に参照できるようになることが望ましいと思います。

ひとまず私が作るコードはrawgit経由で、ここのコードを参照するようにしてみますが、tutorialsと同じ仕組みで載せるのが良さそうに思えます。

satakagi, 2018-09-28:
ちなみに、以下のURLで取ってこれるのはなぜなのでしょう？

https://chirimen.org/chirimen-raspi3/gc/drivers/i2c-PCA9685.js

これができるならrawgitではなくこちらの方が良いでしょうか？

satakagi, 2018-09-30:
https://chirimen.org/chirimen-raspi3/gc/polyfill/polyfill.js

https://chirimen.org/chirimen-raspi3/gc/drivers/i2c-ADS1015.js
https://chirimen.org/chirimen-raspi3/gc/drivers/i2c-ADT7410.js
https://chirimen.org/chirimen-raspi3/gc/drivers/i2c-GP2Y0E03.js
https://chirimen.org/chirimen-raspi3/gc/drivers/i2c-PCA9685.js
https://chirimen.org/chirimen-raspi3/gc/drivers/i2c-S11059.js
https://chirimen.org/chirimen-raspi3/gc/drivers/i2c-VEML6070.js
https://chirimen.org/chirimen-raspi3/gc/drivers/i2c-arduino-steppingMotor.js
https://chirimen.org/chirimen-raspi3/gc/drivers/i2c-canzasi-blink.js
https://chirimen.org/chirimen-raspi3/gc/drivers/i2c-grove-accelerometer.js
https://chirimen.org/chirimen-raspi3/gc/drivers/i2c-grove-gesture.js
https://chirimen.org/chirimen-raspi3/gc/drivers/i2c-grove-light.js
https://chirimen.org/chirimen-raspi3/gc/drivers/i2c-grove-oledDisplay.js
https://chirimen.org/chirimen-raspi3/gc/drivers/i2c-grove-touch.js

dynamis, 2018-10-02:
既存のコードからの読み込み先を書き換える:

- [ ] このリポジトリの gc, _gc でホストしているコード https://github.com/chirimen-oh/chirimen-raspi3/search?q=mz4u.net&unscoped_q=mz4u.net
- [ ] example の jsbin
- [ ] tutorial 本文
- [ ] tutorial の jsbin

ただし、gc, _gc とか tutorial 本文のサンプルなどは、外部リンクじゃなくてセフルホストしている raspi3 上のコードを読み込むようにしなければインターネット接続不要での開発が可能にならないので、そのあたりは注意して書き換える必要がありそう

dynamis, 2018-10-31:
example からのリンクを入れ替えていて気づいたのですが
https://chirimen.org/chirimen-raspi3/...
という project 毎の github pages でのアクセスは
https://chirimen.org/
サイト全体を github pages で運用している場合に限って有効です。

メインサイトに他のシステムを入れるときにはアクセス不能になってしまうという問題があるため専用のドメインを与えてデプロイする形に変えた方が良いかも知れない。

dynamis, 2018-12-17:
note: Netlify の自動デプロイでシェルスクリプトや npm のコマンド類が (依存パッケージのインストール込みで) 使えるので必要なディレクトリ、JS ファイルだけをコピーしたり必要に応じてファイル名を変えたり、minify したバージョンも作ったり、リダイレクトを設定したりしたサイトを作る簡単なスクリプトで書くと比較的シンプルかつ柔軟にリリースできそう。

dynamis, 2019-08-14:
npm 化の副作用対策を慌ててやった時に今後の変更もまだありそうだから r.chirimen.org 化した

https://github.com/chirimen-oh/chirimen-raspi3/issues/68#issuecomment-521204209

closed 2019-08-14 by dynamis