chirimen-oh/chirimen #63: 新しいリリースビルドを作る

dynamis, 2019-02-14:
- [ ] 201811 ビルドをベースにする https://www.raspberrypi.org/downloads/raspbian/
- [ ] などなど

dynamis, 2019-03-22:
Q1 には間に合わなさそうだけど Q2 までには何とかしたい (タイトル変更)

dynamis, 2019-07-26:
201907 の Raspbian Buster ベースにで @WhiteHawk-taka さんに頑張って頂いている

dynamis, 2019-07-26:
pull req https://github.com/chirimen-oh/chirimen-raspi3/pull/67 マージした状態でビルド、リリーステスト
https://github.com/chirimen-oh/chirimen-raspi3/wiki/Release-Checklist
をパスした。

但し、このリポジトリは変更を加えたら release.sh を実行して https://github.com/chirimen-oh/chirimen-raspi3/tree/master/release を更新してもう一度ビルドしてやっとリリースイメージが作れる手順なので、もうすこしざっと見直してから release.sh を実行 (あるいはそもそも gc.zip などはリポジトリに入れずに setup.sh 内で処理させるように変更する) してから月曜にイメージを完成とさせたい。

kou029w, 2019-07-26:
動作確認は問題なくパスしたのですが、いくつか気になった点があったので挙げます
- release/{,_}gc.zip が古い
- 壁紙のバージョンが古い
  - Ver. 2018...
  - 縦幅が720p
- READMEの記載内容がやや古い
  - (Note: default passwd will be changed to rasp) ってありますが、それは 94002a170f8cc6e5fc3ab1b7eac959376da50727 での対応漏れ?

dynamis, 2019-07-26:
これしてリリースしたいチェックリスト:
- [ ] 画面サイズ 720p 固定を解除したので壁紙を 1080 に
- [ ] 壁紙のバージョン記載を自動化したい (cloudinary などで文字を重ねるとか出来るのは知っている)
- [ ] gc.zip と _gc.zip は netlify で自動ビルド
- [ ] release ディレクトリは廃止して setup.sh では netlify のファイルをダウンロード
  - パスは r.chirimen.org で定義しておくと自由に移動可能なのでそのようにする
- [ ] README 更新

あとついでに出来ればだけどこのタイミングで
- [ ] gc 配下のコーディングルール統一
  - チュートリアルの方で統一したものと異なるままになってるので

dynamis, 2019-07-31:
このタイミングで可能な限り突っ込んでいる

- #68 polyfill/driver を npm 管理可能に
- #69 unsupported は消してしまう
- #70 certified/contrib を最新の状況に合わせて整理
  - mbp とかよく使うのを certified に追加していっても良いけどまぁ特に意味はないのでこの程度に
- #71 example の codesandbox 対応
  - 今回これがやりたかった

dynamis, 2019-08-01:
手元にないデバイスを除き一通りの example のテストやリンクチェックなどを行ったが問題なかったのでリリースタグを打った。明日 Upload & 量産予定

https://github.com/chirimen-oh/chirimen-raspi3/releases/tag/20190801

- [ ] test MPU9250
- [ ] test AMG8830

satakagi, 2019-08-01:
両方ともわしが書いたコードですね。まずはOSイメージ作らないと。

satakagi, 2019-08-01:
MPU9250, AMG8830動作確認しました

使用したOSイメージは中くらいのサイズの
Raspbian Buster with desktop　Image with desktop based on Debian Buster
Version:July 2019　Release date:2019-07-10　Kernel version:4.19　Size: 1149 MB
このイメージを使ってRPi3B(無印)で、普通と思われる初期設定を行った後、

wget https://rawgit.com/chirimen-oh/chirimen-raspi3/master/setup.sh
 ./setup.sh
で構築したCHIRIMEN環境で動作確認しています。

kou029w, 2019-08-02:
> Raspbian Buster with desktopというもの（サイズが中間のもの）ですかね？

正しいです

動作確認ありがとうございます :+1:

dynamis, 2019-08-14:
20190801 イメージを UP 済みだが 20190802 イメージを UP できてなかった！

- [ ] upload
- [ ] r.chirimen.org/sdimage 更新