chirimen-oh/chirimen #65: Raspi 互換ボードの検証

dynamis, 2019-05-22:
Raspi 3 でやっとまともに動かせる性能になったから Raspi3 に移植したわけですが、rapi3 ではまだ開発環境としては辛いという声もあったので、Raspberry Pi 3 互換ボードに Debian ベースの OS を入れている場合にも動かせないか試して報告するスレ。

## ASUS Tinkerboard (RK3288)
https://www.asus.com/jp/Single-Board-Computer/Tinker-Board/
動くと小玉さんが行ってた気がします。私も手元に持っているので試してみようかな...

## Rock Pi 4 (RK3399)
model A (無線無し) だと 1GB で $39 と Raspi と十分タメを張れるのに圧倒的な高性能 https://shop.allnetchina.cn/products/rock-pi-4-model-a-board-only-no-wlan-bluetooth-poe?variant=15819370954854

kou029w, 2019-06-06:
## Rock Pi 4 にインストールできない件

### 環境

- Rock Pi 4A DRAM 1GB
- OS:ROCKPI4Debian https://wiki.radxa.com/Rockpi4/Debian
- https://dl.radxa.com/rockpi/images/debian/rockpi4_debian_stretch_lxde_armhf_20181105_2120-gpt.img.gz

### setup.shの実行に関わる問題
- pi → linaro
  - /home/pi → /home/linaro
- /boot/cmdline.txt: No such file or directory
- /etc/xdg/lxsession/LXDE-pi/autostart: No such file or directory
- Package at-spi2-core is not available
- E: パッケージ ttf-kochi-gothic が見つかりません
- E: パッケージ npm が見つかりません
  - https://github.com/nodesource/distributions/blob/master/README.md に従ってインストール可能
- Arduino IDEのインストールに失敗している
- bash: crontab: コマンドが見つかりません
- bash: raspi-config: コマンドが見つかりません
- npm i に失敗
  - makeやg++に依存
    - `sudo apt-get install -y build-essential` で解決
- 日本語入力ができない
- タイムゾーンがUTC
  - JSTにするには `sudo timedatectl set-timezone Asia/Tokyo` で解決
- bash: chromium-browser: コマンドが見つかりません
  - 代わりに chromium として存在している
- tee: /etc/xdg/lxsession/LXDE-pi/autostart: そのようなファイルやディレクトリはありません
- sed: /usr/share/raspi-ui-overrides/applications/lxde-x-www-browser.desktop を読み込めません: そのようなファイルやディレクトリはありません
- tee: /usr/share/raspi-ui-overrides/applications/lxde-x-www-browser.desktop: そのようなファイルやディレクトリはありません
- certutil: unable to open "~/_gc/srv/crt/ca.crt" for reading (-5950, 2).
  - `certutil -A -n org-TripArts -t "TCu,Cu,Tu" -i ~/_gc/srv/crt/ca.crt -d sql:/home/linaro/.pki/nssdb` で解決

<details>
<summary>Package at-spi2-core is not available</summary>

```
Package at-spi2-core is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'at-spi2-core' has no installation candidate
```

</details>
### srv.js 起動に関わる問題
- /home/linaro/_gc/srv/node_modules/raspi-board/dist/index.js:799
  - throw new Error('Unable to parse revision information in /proc/cpuinfo');
    - `RASPI_IO_TEST_MODE=on node srv.js` で無視して起動可能
- /bin/sh: 1: modprobe: not found
  -  /home/linaro/_gc/srv/node_modules/raspi-i2c/dist/index.js:106 コメントアウト (modprobeの使われている行をコメントアウト) すると無視して起動可能

### 余談

https://wiki.radxa.com/Rockpi4/dev/libmraa に従ってRock Pi向けmraaを導入すると比較的簡単にgpioやi2cを触れられるが、カーネルを更新したタイミングでchromiumがクラッシュするようになりました

dynamis, 2019-06-11:
どうやら radxa 公式ビルドのできが悪いが (apt upgrade で Chromium 起動しなくなるし) Armbian のビルドはまともにメンテナンスされてそうなのでやるなら Armbian に CHIRIMEN 環境が追加できるかというトライをした方がよさそう。Ubuntu/Debian 両方あるのをどちらにすべきかは少し悩みどころ。

https://www.armbian.com/rock-pi-4/
https://dl.armbian.com/rockpi-4b/archive/

media script は Ubuntu のみのサポートっぽいのでそちらにすべきかも？
https://forum.armbian.com/topic/9272-development-rk3399-media-script/

dynamis, 2019-10-01:
# Jetson Nano
Raspberry Pi シリーズ以外でちゃんと普及していて重要度の高いボードの筆頭なので検証・対応したいところ

kou029w, 2019-10-01:
## Jetson Nano

setup.shを使ってChirimen for Raspberry Piをインストールするためには、いくつか問題がある (https://github.com/chirimen-oh/chirimen-raspi3/issues/65#issuecomment-499401941 と似た問題)
srv.jsを書き換えてGPIOが動作することまでは確認できた

### 環境
- Jetson Nano Developer Kit JP 4.2.22019/08/26
    - md5sum: 130ece72ef18c7627ba645e94be2e0b5

### GPIOの動作手順

1. package.json,  srv.jsから i2c 周りのコードを取り除く
    - 2bcbc683ffbe8ecc4797c295be38a31acedb195b
2. npm i && sudo node srv.js
3. ブラウザ (chromium-browser) を起動し、GPIO BlinkのExampleページにアクセス
4. ポート番号を適当に書き換える

確認するためのコードは
https://github.com/chirimen-oh/chirimen-raspi3/tree/jetson-nano ブランチに置いています

kou029w, 2019-10-01:
NOTE: raspi-i2cは比較的単純なi2c-busとピンマッピングするためのモジュール (raspi-board, raspi-peripheral) の組み合わせ
RasPi&Raspbian 32Bit環境以外のARMボードの環境に対応するなら、i2c-busモジュールを直接使って書き直す必要があるかも?

kou029w, 2019-10-03:
raspi, raspi-i2cモジュールへの依存を取り除きi2c-busモジュールを直接用いるようにして #82 にて master にマージしました
これによってARM 32bit環境だけでなくi2c-busモジュールがインストール可能なビルドツールチェーンが一通り揃ったLinux環境であればsrv.jsについては問題なく動くようになったんじゃないかなと思います

もちろんRasPi環境でも今までどおり使えると思います
(…がまだ一通り試せていないので不具合無いことを確認したいところです)

dynamis, 2020-12-05:
暫く前に Jetson Nano は 2GB 版で 6820 円という廉価版が出てきた。メモリ半減で価格半減とはね。。。

Raspi 4 も 2GB あればブラウザを動かすのにも足りると検証済みであるし、CPU コアが A53 と A57 という全然性能違うコア (DMIPS だと 2 倍くらい違う) なので 2200 円差くらいなら全然 Jetson Nano 2GB が CHIRIMEN 環境としてはブラウザがより快適に動いて魅力的。もちろん AI など GPU 処理系の場合は更に魅力的。

- Raspi4: Quad Cortex-A53@1.4GHz - 4620 円
  - 別途 CPU ヒートシンクが必須、HDMI が microHDMI でケーブル類も追加で必要
- Jetson Nano: Quad Cortex-A57@1.43GHz - 6820 円
  - 巨大ヒートシンク付きで安心、フル HDMI コネクタ

今年はリモート連携のリレーサーバ系を強化したけど AI サンプルと組み合わせる AI x IoT を JavaScript だけで学習する体験環境として強化していくのにとても良いと思う今日この頃。