chirimen-oh/chirimen #70: 正式サポートデバイスと Contrib デバイスの整理

dynamis, 2019-07-31:
正式サポートデバイスはリリーステストを行うデバイスと定義され、testSet にも含まれるものになっている

https://github.com/chirimen-oh/chirimen-raspi3/tree/master/gc/testSet

逆にこれに含まれないデバイスはテストされないので cotribu 配下に移動させたい

https://github.com/chirimen-oh/chirimen-raspi3/tree/master/gc/i2c/i2c-canzasi-blink
https://github.com/chirimen-oh/chirimen-raspi3/tree/master/gc/i2c/i2c-arduino-steppingMotor

dynamis, 2019-07-31:
- [ ] ディレクトリ移動
- [ ] ブックマーク (ローカルでもリンク) 修正
- [ ] codesandbox redirect 修正

dynamis, 2019-07-31:
i2c detect は逆にメインに入れておくべき

https://github.com/chirimen-oh/chirimen-raspi3/tree/master/gc/contrib/examples/i2c-detect

dynamis, 2019-07-31:
DONE.

dynamis, 2019-07-31:
VL53L0X を certified に格上げ (チュートリアルで扱う距離センサーなので)
https://github.com/chirimen-oh/chirimen-raspi3/tree/master/gc/contrib/examples/i2c-VL53L0X

kou029w, 2019-07-31:
https://github.com/chirimen-oh/chirimen-raspi3/tree/4e9d0749aad70049bb815cf31362f7acd984f8a7/gc/i2c/i2c-VL53L0X に移動しました

https://github.com/chirimen-oh/chirimen-drivers/blob/26bfe3da5fb216bd181f98d66fb1b3920539b5c5/README.md にも記載

closed 2019-07-31 by kou029w