chirimen-oh/chirimen #77: setup.sh で Chromium の Bluetooth 有効化がされていない

dynamis, 2019-08-17:
20190802 ビルドで Chromium の Web Bluetooth が有効化されていないことが判明した。
setup.sh では自動化されていないようだけど、出来ないものでしたっけ？

取りあえず手動で有効化した状態のイメージを作ることは出来るが、自動化したいという issue です。

WhiteHawk-taka, 2019-08-17:
下記オプションにて起動時に有効化されています。
https://github.com/chirimen-oh/chirimen-raspi3/blob/master/setup.sh#L146-L152

WhiteHawk-taka, 2019-08-17:
https://github.com/chirimen-oh/chirimen-raspi3/blob/master/setup.sh#L177-L180
このupgradeで
https://github.com/chirimen-oh/chirimen-raspi3/blob/master/setup.sh#L146-L152
この設定が上書きされておりパネルのブラウザアイコン・アプリケーション一覧のChromiumのBluetoothが無効となっていました。
なので影響がない起動時のChromiumは有効となっていました。
スクリプト作成時は raspberrypi-ui-mods のアップデートがなかったため影響がなかったものの、現在アップデートがあるため問題が発生しました。

WhiteHawk-taka, 2019-08-17:
解決策として L177-L180 のあとに L146-L152 をもってくると解決します。

WhiteHawk-taka, 2019-08-17:
もしくは、自動セットアップに所々問題が発生する raspberrypi-ui-mods のアップグレードは見送る方法では

dynamis, 2019-08-18:
例え出荷時点で回避してもユーザとして受け取ってすぐに update するのとか普通の挙動、アップデートしないでってイメージを配付するのはナシだと思います。

従ってコマンドライン引数以外の方法で設定を自動でオンにする方法を見つけるか、見つかるまでは手動で設定をオンにしてリリースするのが正しいかと思います。

satakagi, 2019-08-18:
パネルのブラウザアイコン・アプリケーション一覧に、WebBTイネーブルにした、「独自の」アプリアイコンを設置できないんですかね？　独自のアプリアイコンも、raspberrypi-ui-mods アップデートかけると消えてしまうならば意味ないですが・・

WhiteHawk-taka, 2019-08-18:
> 例え出荷時点で回避してもユーザとして受け取ってすぐに update するのとか普通の挙動、アップデートしないでってイメージを配付するのはナシだと思います。

ユーザーが普通にupdateしてもapt-mark holdに入れておけばアップデートされません。
例えばcode-ossがそれを使ってバージョン固定しています。
https://github.com/chirimen-oh/chirimen-raspi3/blob/master/setup.sh#L52

kou029w, 2019-08-19:
公式のドキュメントが見当たらなかったためあまりお行儀が良くないhackなのですけど、どうやら chrome://flags を書き換えると "${HOME}/.config/chromium/Local State" に書き込むようなので、あらかじめ Local State ファイルを書き換えるという方法が可能かもしれないです

Local State ファイルは JSON 形式で browser.enabled_labs_experiments オブジェクトの値の配列が変更される:
```json
{"browser":{"enabled_labs_experiments":["enable-experimental-web-platform-features@1"]}}
```

closed 2021-08-06 by kou029w