chirimen-oh/chirimen #78: AMG8833 の配線図で電源が Vin ではなく V3o となっている

dynamis, 2019-08-27:
給電先のピンは Vin であって V3o (V3 out) の誤りの可能性がないでしょうか。

https://github.com/chirimen-oh/chirimen-raspi3/blob/master/gc/contrib/examples/i2c-AMG8833/AMG8833.png

モジュールによっては V3o のない、Vin だけのモジュールもあり (弊社が Ali Express で買ったものもそうでした) Vin に 3.3 or 5V いずれかを接続すれば動作しました。V3o に接続しても駆動する感じでしょうか？

ref: Adafruit の説明ページ
https://learn.adafruit.com/adafruit-amg8833-8x8-thermal-camera-sensor/pinouts
Vin - this is the power pin. Since the sensor uses 3.3V, we have included an onboard voltage regulator that will take 3-5VDC and safely convert it down. To power the board, give it the same power as the logic level of your microcontroller - e.g. for a 5V micro like Arduino, use 5V
3Vo - this is the 3.3V output from the voltage regulator, you can grab up to 100mA from this if you like

kou029w, 2019-08-27:
Adafruit の基板の回路図: https://learn.adafruit.com/adafruit-amg8833-8x8-thermal-camera-sensor/downloads

それぞれの素子のデータシートを調べると

レギュレーター: https://www.mouser.com/datasheet/2/268/mic5225-778308.pdf
AMG88 データシート: http://www.tokaisecurity.net/pdf3/grid-eye_panasonic.pdf

AMG8833自体は定格 VDD 3.3 V ± 0.3 V 4.5 mAで動作可能で、レギュレーターでのドロップ電圧が負荷 50 mA 時最大 300 mV なので、Vin 3.3 Vを下回ったり全体の負荷が50 mA超えるケースを無視すれば基本的にはVinで問題無さそうですね

satakagi, 2019-10-01:
https://learn.adafruit.com/adafruit-amg8833-8x8-thermal-camera-sensor/pinouts
によると、V3oはレギュレータ出力を意図した端子のようですね。
私が持っているブレークアウトボードはAdafruitのものではないので、同じくVinしかないモジュールでVinを使っていますが、この端子に低ドロップのレギュレータが入っているのか確認してみますね。

satakagi, 2019-10-01:
手持ちのブレークアウトボードを確認したところ、[スイッチサイエンスのもの](https://www.switch-science.com/catalog/3395/)でした。基板にはレギュレータが載っていないタイプです。
ということで私の方ではレギュレータなしの直結でしか試していないのですが、うまく動いたということでしたらそっちに直したものに差し替えましょうかね？
一応 dynamis さんお持ちのボードにレギュレータ載っていることを確認しておいていただけますか・