chirimen-oh/chirimen #80: 競合発生時に適切なエラーメッセージを出力すべき

dynamis, 2019-10-01:
他のタブで GPIO/I2C を占有されている場合にその旨のエラーメッセージを出せるはずなのにそうしていないために非常に初学者に分かりにくい。これは Polyfill の実装・配慮不足であり修正したい。

kou029w, 2020-06-19:
- [ ] 環境依存 (CHIRIMEN環境用) かどうかの判定
  - たとえばws connection errorだけではよく分からない

closed 2020-06-19 by kou029w

kou029w, 2020-06-19:
いくつか比較的ポピュラーな問題に関しては日本語のエラーメッセージをDevToolsに表示するようにしました。

接続していないケース:
[image]
競合しているケース:
[image]

PCなど未対応の環境からの接続:
[image]

commit b1a7f8b8 (2020-06-19): console.error 使う & 分かりやすいエラーメッセージ
resolve #80