chirimen-oh/chirimen #87: CHIRIMEN イメージを RasPi 公式サイトに掲載してもらう

dynamis, 2020-01-14:
RasPi 公式 Downloads ページにイメージを "Third Party Operating System Images" として掲載してもらいたい:

https://www.raspberrypi.org/downloads/

しかし掲載申請の手順、審査基準などのドキュメントが見当たらない。誰か知ってる人教えてください。

dynamis, 2020-01-14:
現時点では Rasbian 2019/09 最新イメージベースに対応していて申請しやすい:
https://www.raspberrypi.org/downloads/raspbian/

satakagi, 2020-01-14:
ハイパーリンクを張ってもらう格好のようですので、どのURLにリンクを張ってもらうかを考えたいですね、

dynamis, 2020-01-22:
いまのところ sd image の説明があるのはこちら
https://tutorial.chirimen.org/raspi/sdcard
ですが掲載してもらえるなら chirimen.org/raspi/download 辺りを用意したいですね
ref: https://github.com/chirimen-oh/chirimen-oh.github.io/issues/89#issuecomment-576975935

gurezo, 2021-01-03:
forumsに投稿済み。投稿承認待ち
https://www.raspberrypi.org/forums/index.php

gurezo, 2021-10-09:
beginner カテゴリーで回答がなかったので、Raspberry Pi OSカテゴリーにクロスポスト
https://forums.raspberrypi.com/viewtopic.php?f=66&t=321269

gurezo, 2022-03-14:
サードパーティーを申請する時は、下記リポジトリにissueをたてる
https://github.com/raspberrypi/rpi-imager/issues

kou029w, 2022-03-14:
- CHIRIMEN for Raspberry Pi https://github.com/chirimen-oh/chirimen/releases
- CHIRIMEN Lite https://github.com/chirimen-oh/chirimen-lite/releases

gurezo, 2022-03-14:
TODO
- ダウンロードパージ・リンクを作る
  - https://tutorial.chirimen.org/
  - rpi2,3,4 => https://github.com/chirimen-oh/chirimen/releases
  - pi zero lite => https://github.com/chirimen-oh/chirimen-lite/releases
- 上記ページ・リンク作成後に申請