chirimen-oh/chirimen #100: Raspberry Pi OS 2020.08.20 64bit version problem

gurezo, 2021-01-03:
```
取得:2 http://ftp.jaist.ac.jp/raspbian buster InRelease [15.0 kB]                                                                                
エラー:2 http://ftp.jaist.ac.jp/raspbian buster InRelease                                                                                        
  公開鍵を利用できないため、以下の署名は検証できませんでした: NO_PUBKEY 9165938D90FDDD2E
パッケージリストを読み込んでいます... 完了                                                                                                       
W: GPG エラー: http://ftp.jaist.ac.jp/raspbian buster InRelease: 公開鍵を利用できないため、以下の署名は検証できませんでした: NO_PUBKEY 9165938D90FDDD2E
E: リポジトリ mirror+file:/etc/apt/mirrors.txt buster InRelease は署名されていません。
N: このようなリポジトリから更新を安全に行うことができないので、デフォルトでは更新が無効になっています。
N: リポジトリの作成とユーザ設定の詳細は、apt-secure(8) man ページを参照してください。
raspberrypi-ui-mods はすでに保留に設定されています。
パッケージリストを読み込んでいます... 完了
依存関係ツリーを作成しています                
状態情報を読み取っています... 完了
パッケージ at-spi2-core は使用できませんが、別のパッケージから参照されます。
これは、パッケージが欠落しているか、廃止されたか、または別のソース
からのみ利用可能であることを意味します。

E: パッケージ 'at-spi2-core' にはインストール候補がありません
取得:1 file:/etc/apt/mirrors.txt Mirrorlist [190 B]
取得:2 http://ftp.yz.yamagata-u.ac.jp/pub/linux/raspbian/raspbian buster InRelease [15.0 kB]
エラー:2 http://ftp.yz.yamagata-u.ac.jp/pub/linux/raspbian/raspbian buster InRelease
  公開鍵を利用できないため、以下の署名は検証できませんでした: NO_PUBKEY 9165938D90FDDD2E
ヒット:3 http://archive.raspberrypi.org/debian buster InRelease
パッケージリストを読み込んでいます... 完了
W: GPG エラー: http://ftp.yz.yamagata-u.ac.jp/pub/linux/raspbian/raspbian buster InRelease: 公開鍵を利用できないため、以下の署名は検証できませんでした: NO_PUBKEY 9165938D90FDDD2E
E: リポジトリ mirror+file:/etc/apt/mirrors.txt buster InRelease は署名されていません。
N: このようなリポジトリから更新を安全に行うことができないので、デフォルトでは更新が無効になっています。
N: リポジトリの作成とユーザ設定の詳細は、apt-secure(8) man ページを参照してください。

```

WhiteHawk-taka, 2021-01-03:
現在の setup.sh は以下の p-r の通り 32bit 版をベースに作成されています。
https://github.com/chirimen-oh/chirimen/pull/98

また、 Raspberry Pi 公式サイトの Software ページから通常ダウンロードできないイメージであるため、この issue は保留とします。

gurezo, 2021-01-03:
@WhiteHawk-taka 
有難うございます！
備忘録的に記載しました。

gurezo, 2021-01-17:
2020.08.02 64bit版 の場合、npm, node インストールする前にfull-upgradeする必要がある
```
$ sudo apt-get -y full-upgrade
$ sudo apt-get -y install npm node
```

gurezo, 2021-01-17:
node v14.x 系の場合は、下記コマンドで、npm node インストール完了
```
$ curl -sL https://deb.nodesource.com/setup_14.x | sudo -E bash -
$ sudo apt install nodejs
```

gurezo, 2021-01-18:
64bit 版 VS code のインストール
wget -O /tmp/code.deb 'https://code.visualstudio.com/sha/download?build=stable&os=linux-deb-arm64'

gurezo, 2021-01-18:
試験的シェル
```
#!/bin/bash

# update
sudo apt-get -y update
sudo apt-get -y full-upgrade

# Using Debian, as root
curl -sL https://deb.nodesource.com/setup_14.x | sudo -E bash -
apt-get install -y nodejs

# 各種ツールをインストール
sudo apt-get -y install ttf-kochi-gothic fonts-noto uim uim-mozc apache2 vim emacs libnss3-tools

sudo apt-get -y autoremove
```

gurezo, 2021-02-06:
64bit 

```
pi@raspberrypi:~ $ sudo apt-get -y install ttf-kochi-gothic fonts-noto uim uim-mozc apache2 vim emacs libnss3-tools
パッケージリストを読み込んでいます... 完了
依存関係ツリーを作成しています                
状態情報を読み取っています... 完了
パッケージ vim は使用できませんが、別のパッケージから参照されます。
これは、パッケージが欠落しているか、廃止されたか、または別のソース
からのみ利用可能であることを意味します。

パッケージ fonts-noto は使用できませんが、別のパッケージから参照されます。
これは、パッケージが欠落しているか、廃止されたか、または別のソース
からのみ利用可能であることを意味します。

パッケージ apache2 は使用できませんが、別のパッケージから参照されます。
これは、パッケージが欠落しているか、廃止されたか、または別のソース
からのみ利用可能であることを意味します。

E: パッケージ ttf-kochi-gothic が見つかりません
E: パッケージ 'fonts-noto' にはインストール候補がありません
E: パッケージ uim が見つかりません
E: パッケージ uim-mozc が見つかりません
E: パッケージ 'apache2' にはインストール候補がありません
E: パッケージ 'vim' にはインストール候補がありません
E: パッケージ emacs が見つかりません
E: パッケージ libnss3-tools が見つかりません
pi@raspberrypi:~ $ 
```

gurezo, 2021-02-06:
aptをmirrorで指定している箇所を一時コメント化して問題解消
arduino のセットアップが失敗している
再度確認中

gurezo, 2021-02-06:
```
mkdir: ディレクトリ `/home/pi/.config/chromium/' を作成できません: ファイルが存在します
mkdir: ディレクトリ `/home/pi/.config/chromium/Default/' を作成できません: ファイルが存在します
./setup.x64.sh: 行 165: chromium-browser: コマンドが見つかりません
--2021-02-06 16:28:03--  https://r.chirimen.org/gc.zip
r.chirimen.org (r.chirimen.org) をDNSに問いあわせています... 2400:6180:0:d1::57a:6001, 2400:6180:0:d1::4df:d001, 54.254.38.202, ...
r.chirimen.org (r.chirimen.org)|2400:6180:0:d1::57a:6001|:443 に接続しています... 接続しました。
HTTP による接続要求を送信しました、応答を待っています... 301 Moved Permanently
場所: https://chirimen-gc.netlify.app/gc.zip [続く]
--2021-02-06 16:28:04--  https://chirimen-gc.netlify.app/gc.zip
chirimen-gc.netlify.app (chirimen-gc.netlify.app) をDNSに問いあわせています... 2400:6180:0:d1::684:6001, 2400:6180:0:d0::835:4001, 134.209.106.40, ...
chirimen-gc.netlify.app (chirimen-gc.netlify.app)|2400:6180:0:d1::684:6001|:443 に接続しています... 接続しました。
HTTP による接続要求を送信しました、応答を待っています... 200 OK
長さ: 14369971 (14M) [application/zip]
`gc.zip' に保存中

```

gurezo, 2021-02-06:
```
ln: シンボリックリンク '/usr/local/bin/arduino' の作成に失敗しました: 許可がありません
Adding symlink failed. Hope that's OK. If not then rerun as root with sudo.
 done!
raspberrypi-ui-mods は自動でインストールしたと設定されました。
```

gurezo, 2021-02-06:
2021.02.06 状況整理
- [x] npm, node 
- [x] vscode
- [ ] arduino => NG
- [ ] ブックマーク => NG

gurezo, 2021-02-06:
```
./setup.x64.sh: 行 166: chromium-browser: コマンドが見つかりません

```

gurezo, 2021-02-06:
```
Adding desktop shortcut, menu item and file associations for Arduino IDE...touch: '/home/pi/.local/share/icons/hicolor/.xdg-icon-resource-dummy' に touch できません: そのようなファイルやディレクトリはありません
touch: '/home/pi/.local/share/icons/hicolor/.xdg-icon-resource-dummy' に touch できません: そのようなファイルやディレクトリはありません
touch: '/home/pi/.local/share/icons/hicolor/.xdg-icon-resource-dummy' に touch できません: そのようなファイルやディレクトリはありません
touch: '/home/pi/.local/share/icons/hicolor/.xdg-icon-resource-dummy' に touch できません: そのようなファイルやディレクトリはありません
touch: '/home/pi/.local/share/icons/hicolor/.xdg-icon-resource-dummy' に touch できません: そのようなファイルやディレクトリはありません
touch: '/home/pi/.local/share/icons/hicolor/.xdg-icon-resource-dummy' に touch できません: そのようなファイルやディレクトリはありません
touch: '/home/pi/.local/share/icons/hicolor/.xdg-icon-resource-dummy' に touch できません: そのようなファイルやディレクトリはありません
touch: '/home/pi/.local/share/icons/hicolor/.xdg-icon-resource-dummy' に touch できません: そのようなファイルやディレクトリはありません
touch: '/home/pi/.local/share/icons/hicolor/.xdg-icon-resource-dummy' に touch できません: そのようなファイルやディレクトリはありません
touch: '/home/pi/.local/share/icons/hicolor/.xdg-icon-resource-dummy' に touch できません: そのようなファイルやディレクトリはありません
touch: '/home/pi/.local/share/icons/hicolor/.xdg-icon-resource-dummy' に touch できません: そのようなファイルやディレクトリはありません
touch: '/home/pi/.local/share/icons/hicolor/.xdg-icon-resource-dummy' に touch できません: そのようなファイルやディレクトリはありません
touch: '/home/pi/.local/share/icons/hicolor/.xdg-icon-resource-dummy' に touch できません: そのようなファイルやディレクトリはありません
touch: '/home/pi/.local/share/icons/hicolor/.xdg-icon-resource-dummy' に touch できません: そのようなファイルやディレクトリはありません
touch: '/home/pi/.local/share/icons/hicolor/.xdg-icon-resource-dummy' に touch できません: そのようなファイルやディレクトリはありません
touch: '/home/pi/.local/share/icons/hicolor/.xdg-icon-resource-dummy' に touch できません: そのようなファイルやディレクトリはありません
touch: '/home/pi/.local/share/icons/hicolor/.xdg-icon-resource-dummy' に touch できません: そのようなファイルやディレクトリはありません
touch: '/home/pi/.local/share/icons/hicolor/.xdg-icon-resource-dummy' に touch できません: そのようなファイルやディレクトリはありません

```

gurezo, 2021-02-06:
```
ln: シンボリックリンク '/usr/local/bin/arduino' の作成に失敗しました: 許可がありません
Adding symlink failed. Hope that's OK. If not then rerun as root with sudo.
 done!

```

gurezo, 2021-06-05:
2021-05-28版が出たので、再チャレンジ
https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2021-05-28/

gurezo, 2021-07-29:
ttf-kochi-gothic が見つからない
それ以外は32 bit と同じで行ける？

gurezo, 2021-07-29:
sudo apt-cache search font japanese | grep ttf-kochi-gothic 
上記コマンドで確認

gurezo, 2021-07-29:
> sudo apt-cache search font japanese | grep ttf-kochi-gothic

検索結果ゼロ。無くなった？

gurezo, 2021-07-29:
廃止予定とのこと
https://packages.debian.org/ja/jessie/ttf-kochi-gothic

gurezo, 2021-08-10:
2021.08.10 状況整理
- [x] npm, node => 32bit 方式
- [ ] vscode
  - [x] apt-get install OK 
  - [ ] 64bit版かどうか要確認
- [ ] arduino 
  - [x] apt-get install OK 
  - [ ] ショートカットアイコン が複数できる
  - [ ] 64bit版かどうか要確認
- [ ] ブックマーク
  - [ ] チュートリアル
    - [ ] 未確認
  - [ ] ヘルプ
    - [ ] 未確認
  - [ ] Top
    - [ ] 未確認
  - [ ] Example
    - [ ] 未確認
  - [ ] JS Bin
    - [ ] 未確認
  - [ ] JSFiddle
    - [ ] 未確認
  - [ ] codesandbox
    - [ ] 未確認
  - [ ] CHIRIMEN サイト
    - [ ] 未確認
- [ ] apt の sourcelist
  - ※暫定でsource.list指定をコメント化して問題回避中
- [ ] apache が起動していない

gurezo, 2021-08-10:
64bit 指定
```
# example
 sudo apt-get install libgtk-3-0:amd64
```

gurezo, 2021-08-16:
https://www.labohyt.net/blog/server/post-4721/

gurezo, 2021-08-16:
5.4. dpkg を用いたパッケージの操作
https://debian-handbook.info/browse/ja-JP/stable/sect.manipulating-packages-with-dpkg.html

gurezo, 2021-08-16:
セットアップログ
[x64.txt](https://github.com/chirimen-oh/chirimen/files/6989985/x64.txt)

gurezo, 2021-08-16:
### VS Code

```
mkdir: cannot create directory ‘/home/pi/.config/chromium/’: File exists
mkdir: cannot create directory ‘/home/pi/.config/chromium/Default/’: File exists
./setup.x64.sh: line 169: chromium-browser: command not found

```

### apache2 or SSL
```
Enabling site vhost-ssl.
To activate the new configuration, you need to run:
  systemctl reload apache2
Considering dependency setenvif for ssl:
Module setenvif already enabled
Considering dependency mime for ssl:
Module mime already enabled
Considering dependency socache_shmcb for ssl:
Enabling module socache_shmcb.
Enabling module ssl.
See /usr/share/doc/apache2/README.Debian.gz on how to configure SSL and create self-signed certificates.
To activate the new configuration, you need to run:
  systemctl restart apache2
Notice: Trust flag u is set automatically if the private key is present.
]0;pi@raspberrypi: ~/chirimen[01;32mpi@raspberrypi[00m:[01;34m~/chirimen $[00m exit

```

gurezo, 2021-08-16:
### pi-bluetooth
```
Setting up pi-bluetooth (0.1.17) ...
Job for hciuart.service failed because the control process exited with error code.
See "systemctl status hciuart.service" and "journalctl -xe" for details.
Job for hciuart.service failed because the control process exited with error code.
See "systemctl status hciuart.service" and "journalctl -xe" for details.

```

gurezo, 2021-08-16:
mirror site 
https://www.debian.org/mirror/list.ja.html

gurezo, 2021-08-16:
```
W: Skipping acquire of configured file 'rpi/binary-armhf/Packages' as repository 'mirror+file:/etc/apt/mirrors.txt buster InRelease' doesn't have the component 'rpi' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'rpi/binary-arm64/Packages' as repository 'mirror+file:/etc/apt/mirrors.txt buster InRelease' doesn't have the component 'rpi' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'rpi/i18n/Translation-en' as repository 'mirror+file:/etc/apt/mirrors.txt buster InRelease' doesn't have the component 'rpi' (component misspelt in sources.list?)
```

https://askubuntu.com/questions/741410/skipping-acquire-of-configured-file-main-binary-i386-packages-as-repository-x

gurezo, 2021-08-16:
2021.08.16 状況整理
- [x] フォント問題 => 32bit 方式 https://github.com/chirimen-oh/chirimen/issues/109#issuecomment-893095558
- [x] npm, node => 32bit 方式
- [ ] vscode
  - [x] apt-get install OK 
  - [x] 64bit版かどうか要確認 =>64bit
  - [ ] デスクトップにショートカットができていない
  - [ ] 設定に失敗している可能性あり
- [x] arduino 
  - [x] apt-get install OK 
  - [x] ショートカットアイコン が複数できる => 旧セットアップを削除で解決
  - [x] 64bit版かどうか要確認=>64bit
- [x] ブックマーク
  - [x] チュートリアル
  - [x] ヘルプ
  - [x] Top
  - [x] Example
  - [x] JS Bin
  - [x] JSFiddle
  - [x] codesandbox
  - [x] CHIRIMEN サイト
- [x] 壁紙
- [x] apt の sourcelist => [修正](d3391d0)
  - ~~※暫定でsource.list指定をコメント化して問題回避中~~
- [ ] ~~apache が起動していない~~ トップページが開かない

gurezo, 2021-08-16:
セットアップログ
[x64-setup.txt](https://github.com/chirimen-oh/chirimen/files/6992761/x64-setup.txt)

gurezo, 2021-11-08:
2021-11-08リリース
Index of /raspios_arm64/images/raspios_arm64-2021-11-08
https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2021-11-08/

gurezo, 2022-02-04:
Raspberry Pi OS 2022.01.28 64bit 版が出たので、一旦close

#118 へ引き継ぎ

closed 2022-02-04 by gurezo