chirimen-oh/chirimen #104: リリース作業

kou029w, 2021-01-05:
- [ ] [Link Checker](https://validator.w3.org/checklink) にて https://chirimen.org/chirimen/gc/top/examples/ 内のリンクが壊れていないことを確認しておく…確認中
- [ ] Raspberry Pi OS 32-bit 書き込む…書き込み中
- [ ] イメージ作成
  - setup.sh 実行する
  - 検索エンジンをGoogleに変更
  - イメージ吸出しとシュリンク
- [ ] テスト https://github.com/chirimen-oh/chirimen/wiki/Release-Checklist

kou029w, 2021-01-05:
イメージ作りました
Lチカすることだけ手元の機体で確認しています
問題のあったRaspberry Pi 4Bの機体でも問題なく動くかどうかなど未確認です

kou029w, 2021-01-05:
https://github.com/chirimen-oh/chirimen/releases/tag/20210105 にダウンロードできるリンクを記載しています。
現状権限の都合でGoogle Driveにてホスティングしています。
実機等で確認してもらって問題なければdownload.chirimen.orgにホスティングしてもらえますか? >  @dynamis 
あるいはdownload.chirimen.orgへの権限等もらえればホスティングに関しては対応できるかもと思います。
よろしくお願いします。

dynamis, 2021-01-06:
## フルテストメモ用チェックリスト
- GPIO
  - [x] GPIO blinkAll
  - [x] GPIO buttonAll
- I2C 基礎
  - [x] I2C detect
  - [x] SHT30
  - [ ] ADT7410
    - 一見正常動作するが 32 ℃くらいを越えると -479 ℃ などという温度になる
- I2C
  - [x] ADS1015
  - [x] ADS1115
  - [ ] AMG8833
  - [x] APDS9960
  - [x] BH1750
  - [x] BME280
  - [x] BMP180
  - [x] BMP280
  - [ ] CCS811
  - [ ] INA219
  - [x] MLX90614
  - [x] MPU6050
    - データ校正用の温度も得ていることを example に追記する
  - [x] MPU9250 (MPU6500 + AK8963)
    - demo/csb は問題なし。jsbin サンプルではエラーになる (原因未確認)
    - todo: temp 行があるが MPU6050 と違い温度は得てないので example から削除
  - [ ] NEOPIXEL-I2C
  - [x] PAJ7620U2 (= Grove Gesture)
  - [ ] PCA9685
  - [ ] PCF8591
  - [x] S11059
  - [ ] TCA9548A
  - [ ] TCS34725
  - [x] VEML6070
  - [x] VL53L0X
  - [x] VL53L1X
- I2C (other)
  - [ ] ARDUINO-STEPPING-MOTOR
  - [ ] CANZASI
  - [ ] GP2YOE03
  - [ ] SEESAW
- Grove
  - [x] Accelerometer
  - [x] Light
  - [x] Gesture (PAJ7620U2)
  - [x] Touch
  - [x] Oled Display (SSD1308)

dynamis, 2021-01-06:
ADT7410 のドライバの氷点下温度の判断ビット位置の間違いはドライバ側のバグとして修正された:

https://github.com/chirimen-oh/chirimen-drivers/pull/103

dynamis, 2021-01-06:
> 実機等で確認してもらって問題なければdownload.chirimen.orgにホスティングしてもらえますか? > @dynamis
> あるいはdownload.chirimen.orgへの権限等もらえればホスティングに関しては対応できるかもと思います。

サーバ直接のアクセス権は持ってないですが、ミラー元となる GDrive がこちらです:

https://drive.google.com/drive/u/0/folders/1s8Aa0XCrJWxiHFn_UFBCHThOSgj6meCp

入れたら rclone で 10 分ごとに sync してくれる設定とのこと。フォルダの書き込み共有権限付与しました。

kou029w, 2021-01-06:
https://github.com/chirimen-oh/chirimen/releases/tag/20210106 デプロイしました > @dynamis

dynamis, 2021-01-06:
20210106 で大丈夫そうですね。一応重要サブセットテスト流します:

## 確認
- [ ] 新旧 RasPi4 リビジョンでの起動
- [x] 起動時自動ブラウザ起動
- [x] 壁紙
- [x] ブックマーク
- [x] reset.sh

## フルテストメモ用チェックリスト
- GPIO
  - [x] GPIO blinkAll
  - [x] GPIO buttonAll
- I2C 基礎
  - [x] I2C detect
  - [x] SHT30
  - [x] ADT7410
- IoT Tinkering Sensor Box
  - [ ] ADS1115
  - [ ] AMG8833
  - [ ] BH1750
  - [ ] BME280
  - [ ] MLX90614
  - [ ] Neopixcel Driver + WS2812
  - [ ] PCA9685 + SG90
  - [ ] VL53L0X

dynamis, 2021-01-06:
raspi3 じゃなく raspi になったので /release/raspberry_pi_3/ ではなく /release/raspi/ ディレクトリに移動させますね。

dynamis, 2021-01-06:
今回のリリースに change notes も書き足しました。ということで一応リリース完了かな！

closed 2021-01-06 by dynamis