chirimen-oh/chirimen #109: 2021-08 リリース作業

kou029w, 2021-07-16:
関連: #104, #63, #45

リリース作業を実施する

gurezo, 2021-07-29:
@kou029w 
フォント情報で、共有します。
ttf-kochi-gothicは、廃止予定とのことです。
https://github.com/chirimen-oh/chirimen/issues/100#issuecomment-889011706

64bit 版を試している時に判明しました。

gurezo, 2021-07-29:
@kou029w 
下記箇所で、apt-get失敗してキチンとインストールされないので、ご注意ください
https://github.com/chirimen-oh/chirimen/blob/master/setup.sh#L62

kou029w, 2021-08-05:
@gurezo 
共有ありがとうございます。既にfonts-notoが含まれていますがttf-kochi-gothicも含め日本語フォントとしては[fonts-noto-cjk](https://packages.debian.org/stable/fonts/fonts-noto-cjk)あたりに置き換えたほうが良さそうですね。

---

そういえばフォントのインストールは行っていますがブラウザーのデフォルトのフォントとしての設定やデスクトップのデフォルトのフォントの設定は行われていないのでしょうかね…? もしかしたらやったほうがよいのかな。

kou029w, 2021-08-05:
https://github.com/chirimen-oh/chirimen/blob/f935a561825f556097beb1884682b88d8fd8a4e0/gc/top/examples/index.html#L1075 
`<a href="https://www.amazon.co.jp/dp/B07CHDBW38/">マトリクス</a>` リンク切れ

kou029w, 2021-08-05:
VSCodeのインストールは下記でOK
```sh
sudo apt install code
```
参考文献: https://qiita.com/hikaru__m/items/09ac6b38a42e53074bbf

kou029w, 2021-08-05:
Arduino IDEも下記で済むのではないだろうか
```sh
sudo apt install arduino
```
https://packages.debian.org/buster/arduino

kou029w, 2021-08-05:
Node.js は npm パッケージ n を使うより APT リポジトリ NodeSource を使うほうが管理の単純さの点でメリットがあると考える
n そのものを動かすためのランタイムの Node.js に加えて n によって管理される Node.js の少なくとも 2 つをメンテナンスする必要があるが、NodeSource を使いAPTリポジトリに統一することによってその必要が無くなる

```sh
curl -sL https://deb.nodesource.com/setup_lts.x | sudo -E bash -
sudo apt-get install -y nodejs
```

https://github.com/nodesource/distributions#installation-instructions

ここではLTS (2021-08-05 現在 v14)のインストール方法を示したが[gpioパッケージとの互換性に問題があるらしい](https://github.com/chirimen-oh/chirimen/pull/98)のでこれはあとで確認してみる

kou029w, 2021-08-05:
> `<a href="https://www.amazon.co.jp/dp/B07CHDBW38/">マトリクス</a>` リンク切れ

マトリックス https://www.amazon.co.jp/dp/B074Z221XV
リング https://www.switch-science.com/catalog/1537/

kou029w, 2021-08-05:
手動でやる必要がありそうな事項:

- Wifiをオン
- デスクトップ環境
  - 必要に応じてフォントの変更
- ブラウザー周り
  - Web Bluetooth API
  - ツールバーの表示
  - 検索エンジンをGoogleに変更
  - 必要に応じてフォントの変更

kou029w, 2021-08-05:
> Arduino IDEも下記で済むのではないだろうか
> 
> ```sh
> sudo apt install arduino
> ```
> 
> https://packages.debian.org/buster/arduino

済む。 (2021-05-07 Raspberry Pi OS with desktop Kernel 5.10)

kou029w, 2021-08-05:
> ここではLTS (2021-08-05 現在 v14)のインストール方法を示したが[gpioパッケージとの互換性に問題があるらしい](https://github.com/chirimen-oh/chirimen/pull/98)のでこれはあとで確認してみる

再現した。

https://github.com/chirimen-oh/chirimen/blob/aa74b01e86c5a58acf14739f5980f84b8deac0eb/_gc/srv/srv.js#L485

ここのportnumにnumberを許容しないようになった。

https://github.com/chirimen-oh/chirimen/blob/aa74b01e86c5a58acf14739f5980f84b8deac0eb/_gc/srv/srv.js#L508

ここのvalueにいかなる値も許容しないようになった。number以外を与えるとgpioによってnumberに変換されエラーが発報する。
[onoff](https://www.npmjs.com/package/onoff) あたりに置き換えたほうが良いと考える。

kou029w, 2021-08-06:
> [onoff](https://www.npmjs.com/package/onoff) あたりに置き換えたほうが良いと考える。

#110 にて対応

kou029w, 2021-08-10:
https://github.com/kou029w/chirimen-os/tree/full をローカル環境で実行しビルド中…。

```sh
git clone https://github.com/kou029w/chirimen-os.git
cd chirimen-os
git switch full
./build-docker.sh
```

kou029w, 2021-08-10:
> https://github.com/kou029w/chirimen-os/tree/full をローカル環境で実行しビルド中…。

これはいくつかのコマンドの実行に失敗したため断念した。
公式のRaspberry Pi OS Full版のイメージをもとにしてRaspberry Pi 4実機でセットアップする方針にする。

pi-genを使用したローカル環境でのセットアップで失敗したもの:

- chromiumの起動後のセットアップ
  - 証明書
  - ホームページ
- 壁紙
- Wifiの地域の設定

kou029w, 2021-08-12:
https://github.com/chirimen-oh/chirimen/releases/tag/20210812 にてpre-release完了。
明日手元にあるいくつかのデバイスを使ってテストを行う予定。

- [ ] GPIO
- [ ] SHT30
- [ ] ADT7410
- [ ] VL53L0X
- [ ] BMP280

kou029w, 2021-08-13:
私の手元にあるデバイスを試した範囲では問題無さそうに思う。

kou029w, 2021-08-13:
https://github.com/chirimen-oh/chirimen/releases/tag/20210812 リリース :tada:

closed 2021-08-13 by kou029w