chirimen-oh/chirimen #114: webGPIO onchange仕様が異なる

satakagi, 2021-09-07:
以下参照
https://github.com/chirimen-oh/node-web-gpio/issues/33

http://browserobo.github.io/WebGPIO/

satakagi, 2021-09-07:
雑ですがこんな感じですか
diff polyfill_patched.js polyfill.js

```javascript
255,269d254
<     addEventListener: function(ename,cbf){
<       if ( ename != "change" || typeof cbf != "function"){
<               return;
<       }
<       bone.registerEvent(0x14, this.portNumber, (buf) => {
<         infoLog("change event");
<         cbf({port:this,value:buf[5]});
<       });
<     },
<     removeEventListener: function(ename,cbf){
<       if ( ename != "change" || typeof cbf != "function"){
<         return;
<       }
<       bone.removeEvent(0x14, this.portNumber);
<     },
282c267
<                       this.onchange({port:this,value:buf[5]});
---
>               this.onchange(buf[5]);
```

satakagi, 2021-09-07:
あ、bone.registerEventは一個しか登録できない・・　雑過ぎか。

gurezo, 2025-04-26:
cf. https://github.com/chirimen-oh/meeting/blob/master/2025/meeting-2025.04.21.md

- Chirimen Desktop 版保留（塩漬け）
- Chirimen Desktop 版のニーズが出たら、相談する
- Chirimen Desktop 版を作り直す時に修正する

closed 2025-04-26 by gurezo